Seeed-Studio/Seeed_GFX #18: Guru meditation error XIAO ESP32S3

bmiles, 2025-05-02:
I see the below error looping in the serial monitor after following this guide: https://wiki.seeedstudio.com/XIAO-eInk-Expansion-Board/

Using Seeed ESP32-S3, XIAO e-ink expansion board and XIAO 2.9 inch e-ink display
Board manager → 2.0.13
IDE → Arduino
Seeed_Arduino_LCD Lib version → v2.0.3

I made the recommended edit to User_Setup_Select.h

I have this issue with the lib examples: Shape.ino, Clock.ino, Bitmap.ino

```
19:42:06.771 -> Build:Mar 27 2021
19:42:06.771 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
19:42:06.771 -> Saved PC:0x40379032
19:42:06.771 -> SPIWP:0xee
19:42:06.771 -> mode:DIO, clock div:1
19:42:06.771 -> load:0x3fce3818,len:0x109c
19:42:06.771 -> load:0x403c9700,len:0x4
19:42:06.771 -> load:0x403c9704,len:0xb50
19:42:06.771 -> load:0x403cc700,len:0x2fd0
19:42:06.771 -> entry 0x403c98ac
19:42:08.896 -> Start
19:42:08.896 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
19:42:08.896 -> 
19:42:08.896 -> Core  1 register dump:
19:42:08.896 -> PC      : 0x4200321e  PS      : 0x00060430  A0      : 0x820032ac  A1      : 0x3fcebde0  
19:42:08.896 -> A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x00000009  
19:42:08.896 -> A6      : 0x000000ff  A7      : 0x3fcec3bb  A8      : 0x08000000  A9      : 0x3fcebdb0  
19:42:08.896 -> A10     : 0x3fc9521c  A11     : 0x00000001  A12     : 0x00989680  A13     : 0xffffffff  
19:42:08.928 -> A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
19:42:08.928 -> EXCVADDR: 0x00000010  LBEG    : 0x42009a60  LEND    : 0x42009aa7  LCOUNT  : 0x00000002  
19:42:08.928 -> 
19:42:08.928 -> 
19:42:08.928 -> Backtrace: 0x4200321b:0x3fcebde0 0x420032a9:0x3fcebe10 0x42004fa2:0x3fcebe30 0x42001bb6:0x3fcebe50 0x4200a622:0x3fcebe70 0x4037d576:0x3fcebe90
19:42:08.928 -> 
19:42:08.928 -> 
19:42:08.928 -> 
19:42:08.928 -> 
19:42:08.928 -> ELF file SHA256: d4199802fd15d4d7
19:42:09.060 -> 
19:42:09.060 -> Rebooting...
```

LynnL4, 2025-05-09:
Hi,
Which version of the ESP32 Arduino Core are you using? Please try updating to the latest version.

closed 2025-05-30 by github-actions[bot]

okwes, 2025-07-28:
Hello. I am having this same issue, this time with the 7.5in monochrome e ink display and the XAIO esp32-s3. Likewise this was after changeing the User_Setup_Select.h

likewise tested with Shape.ino and Clock.ino examples

Board manager: 3.3.0 
IDE: Arduino
Seeed_Arduino_LCD Lib version:  v2.0.3

Error when running from serial log:
```log
09:47:47.826 -> ESP-ROM:esp32s3-20210327
09:47:47.826 -> Build:Mar 27 2021
09:47:47.826 -> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
09:47:47.826 -> Saved PC:0x4037837a
09:47:47.826 -> SPIWP:0xee
09:47:47.826 -> mode:DIO, clock div:1
09:47:47.826 -> load:0x3fce2820,len:0x1180
09:47:47.826 -> load:0x403c8700,len:0xc2c
09:47:47.826 -> load:0x403cb700,len:0x311c
09:47:47.826 -> entry 0x403c88b8
09:47:47.923 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
09:47:47.923 -> 
09:47:47.923 -> Core  1 register dump:
09:47:47.923 -> PC      : 0x42003c84  PS      : 0x00060430  A0      : 0x82003d10  A1      : 0x3fcebde0  
09:47:47.955 -> A2      : 0x3fc96b40  A3      : 0x00000000  A4      : 0x00000008  A5      : 0x00000009  
09:47:47.955 -> A6      : 0x000000ff  A7      : 0x60004000  A8      : 0x00000010  A9      : 0x08000000  
09:47:47.955 -> A10     : 0x3fc96cc0  A11     : 0x00989680  A12     : 0x00000001  A13     : 0x00000000  
09:47:47.987 -> A14     : 0x00989680  A15     : 0x3fcebdb8  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
09:47:47.987 -> EXCVADDR: 0x00000010  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
09:47:47.987 -> 
09:47:47.987 -> 
09:47:47.987 -> Backtrace: 0x42003c81:0x3fcebde0 0x42003d0d:0x3fcebe10 0x42005cbd:0x3fcebe30 0x42001ed2:0x3fcebe50 0x4200b2fa:0x3fcebe80 0x4037b40d:0x3fcebea0
09:47:47.987 -> 
09:47:47.987 -> 
09:47:47.987 -> 
09:47:47.987 -> 
09:47:47.987 -> ELF file SHA256: cbff143b5
09:47:48.117 -> 
09:47:48.117 -> Rebooting...
```

Verbose compilation log
```log
FQBN: esp32:esp32:XIAO_ESP32S3
Using board 'XIAO_ESP32S3' from platform in folder: /Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0
Using core 'esp32' from platform in folder: /Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0

/usr/bin/env bash -c "[ ! -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/partitions.csv ] || cp -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/partitions.csv \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/partitions.csv ] || [ ! -f \"/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0/variants/XIAO_ESP32S3\"/partitions.csv ] || cp \"/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0/variants/XIAO_ESP32S3\"/partitions.csv \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/partitions.csv ] || cp \"/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0\"/tools/partitions/default_8MB.csv \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/bootloader.bin ] && cp -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/bootloader.bin \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/Clock.ino.bootloader.bin || ( [ -f \"/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0/variants/XIAO_ESP32S3\"/bootloader.bin ] && cp \"/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0/variants/XIAO_ESP32S3\"/bootloader.bin \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/Clock.ino.bootloader.bin || \"/Users/wesley/Library/Arduino15/packages/esp32/tools/esptool_py/5.0.0\"/esptool --chip esp32s3 elf2image --flash-mode dio --flash-freq 80m --flash-size 8MB -o \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/Clock.ino.bootloader.bin \"/Users/wesley/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.5-b66b5448-v1/esp32s3\"/bin/bootloader_qio_80m.elf )"
esptool v5.0.0
Creating ESP32S3 image...
Merged 2 ELF sections.
Successfully created ESP32S3 image.
/usr/bin/env bash -c "[ ! -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/build_opt.h ] || cp -f \"/private/var/folders/4k/f1xcy05n4fs87njwdyvqxm780000gn/T/.arduinoIDE-unsaved2025628-50594-972082.yl1w/Clock\"/build_opt.h \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/build_opt.h"
/usr/bin/env bash -c "[ -f \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/build_opt.h ] || : > \"/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4\"/build_opt.h"
/usr/bin/env bash -c ": > '/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4/file_opts'"
[... cut ...]
"/Users/wesley/Library/Arduino15/packages/esp32/tools/esptool_py/5.0.0/esptool" --chip esp32s3 --port "/dev/cu.usbmodem2101" --baud 921600  --before default-reset --after hard-reset write-flash  -z --flash-mode keep --flash-freq keep --flash-size keep 0x0 "/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4/Clock.ino.bootloader.bin" 0x8000 "/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4/Clock.ino.partitions.bin" 0xe000 "/Users/wesley/Library/Arduino15/packages/esp32/hardware/esp32/3.3.0/tools/partitions/boot_app0.bin" 0x10000 "/Users/wesley/Library/Caches/arduino/sketches/C5382408B7D0D8A82CCF39CCB2F134D4/Clock.ino.bin" 
```

okwes, 2025-07-28:
Hello, I was looking though the code and realized a commit was made since the last release that fixed this issue, and I tried it and it worked. Since the last release was half a year ago maybe it would be a good make another release or mention in the getting started to just clone from master:)